sparkfun/Fritzing_Parts #6: Is this the official SPE repo for fritzing parts?

liudr, 2015-06-22:
Is this the official SPE repo for fritzing parts?

I've not been up to date with SPE fritzing library for about a year. Was looking for a part for this interrupter breakout board but didn't find fritzing part:

https://www.sparkfun.com/products/9322

I was making some workshop slides BTW.

Thanks.

bhuang10000, 2015-06-22:
Hi liudr - 

Yep, this is the most up-to-date repo for fritzing parts. We have a few folks here that are pretty fast at building up new parts in Fritzing. If you have this part made, please feel free to contribute back to this repo with a pull request. 

Thanks!

liudr, 2015-06-24:
Thanks Brian! I've not built it yet. I did build a couple of parts in the past but forgot how. I'll give it a try though. Having a Fritzing diagram is a lot more clear than taking a photo of even the neatest breadboard. BTW, I'm using this part for the AAPT summer meeting workshop. I'll see you there then :)

closed 2015-06-25 by ToniCorinne

karma0, 2023-04-06:
Has anyone created the photo interrupter board?  I'm not finding it in the repo, but need it.