sparkfun/Fritzing_Parts #12: Where to put these files

dagwieers, 2016-04-14:
It would help if the **README** stated where exactly I have to put these directories and files so Fritzing picks them op. I tried **parts/contrib** and **parts/user,** to no avail.

A small set of hints would be much appreciated.

aknoerig, 2016-04-15:
Take a look at the freshly refined instructions at [https://github.com/fritzing/fritzing-parts/blob/master/CONTRIBUTING.md](https://github.com/fritzing/fritzing-parts/blob/master/CONTRIBUTING.md).
Hope this answers your question, since it seems to be fritzing generic rather than Sparkfun specific?

dagwieers, 2016-04-15:
Thanks and true.

But users ending up on your website will also have to perform this operation.
So if I may suggest to add the instructions and a link to the README.md of this project ?
(When looking for answers I came across many inquiries and no concrete answers like yours)

Thanks again !

aknoerig, 2016-04-15:
@ToniCorinne, could you take care of this?

ToniCorinne, 2016-04-19:
Added - thanks!

closed 2016-04-19 by ToniCorinne

dagwieers, 2016-04-20:
Thanks a lot !