sparkfun/Fritzing_Parts #16: no .fzbz file to import

MarcusWolschon, 2017-05-19:
There is no .fzbz file to import and the README.md doesn't state how exactly to import this part library into Fritzing.

MarcusWolschon, 2017-05-19:
The linked documentation is idiotic. You don't git checkout anything and then move all part of it somewhere outside the checkout folder. You'd never be able to do a git pull because you have uncommited local changes (deleting all those files) that you can't possibly merge.

MarcusWolschon, 2018-08-16:
After 1.5 years? How should I still know. 
Probably the Sparkfun AutoDriver V11 and V13 boards to recreate a setup+pinout and render graphics  for user documentation.

loricrotser, 2018-08-16:
Apologies for the long reply time; this is newly assigned to me. 

I'll see what I can do about getting you that autodriver board in fritzing. I'm assuming you mean the L6470 - let me know if that's not correct. 

Thanks

MarcusWolschon, 2018-08-17:
Yes, that's the chip on these board revisions.

loricrotser, 2018-08-17:
Initial checkin of v13:
[https://github.com/sparkfun/Fritzing_Parts/blob/master/products/13752_sfe_autodriver_stepper_motor_v13.fzpz](https://github.com/sparkfun/Fritzing_Parts/blob/master/products/13752_sfe_autodriver_stepper_motor_v13.fzpz)

V10 has been retired - I'll be working on that one next week. 
Thank you for your patience!

loricrotser, 2018-08-20:
I've just uploaded the initial checkin for v10 of the autodriver: 
[https://github.com/sparkfun/Fritzing_Parts/blob/master/products/11611_sfe_autodriver_stepper_motor_v10.fzpz](https://github.com/sparkfun/Fritzing_Parts/blob/master/products/11611_sfe_autodriver_stepper_motor_v10.fzpz)

Double clicking on the .fzpz will open it in fritzing and import the part. 

I've also logged an internal ticket to look into providing bins for customer fritzing parts. 

Thank you!

closed 2018-08-20 by loricrotser