sparkfun/Fritzing_Parts #28: Suggestion: Add instructions on how to import the parts

sbrl, 2019-06-15:
As it stands, I'm unsure how to import these into Fritzing. I can do them one at a time through the GUI, but that would take absolutely _ages_.

Is there a way to import them all at once?

BobGreenwade, 2019-11-02:
I echo this concern. I'm a little surprised it's not there already. Adafruit's library is this simple; why not SparkFun's?

loricrotser, 2019-11-08:
We're working on this. Traditionally, we've submit tickets to Fritzing to have them add parts to our existing bin that is shipped with the executable. That hadn't been updated in quite some time, so we're looking at delivering fzbz files here to make life easier. 

It's coming - hang in there!