sparkfun/Fritzing_Parts #30: Missing Fritzing file for SparkFun Arduino ProtoShield (DEV-13819/ DEV-13820)

dltj, 2020-05-23:
It would be really useful to have a Fritzing file for the [SparkFun Arduino ProtoShield](https://www.sparkfun.com/products/13819).  Being new to PCB design, I would find it reassuring to know that the parts I've laid out on the board match the design schematic in Fritzing's Breadboard/Schematic back-and-forth view.

bboyho, 2020-05-28:
Hi,

I remember we tried to make the part from the Eagle files when it originally came out There was something funky about the design with the protosnap part causing it to not render correctly in Fritzing when using the Eagle-to-Fritzing tool.

I'm not sure of the timeline but I can try to take a look at this again to see what we can do to make a part.

dltj, 2020-05-28:
Thank you -- much appreciated.