sparkfun/Fritzing_Parts #34: Error when importing fritzing library

wegunterjr, 2023-08-02:
Recreate:
I either open the fzbz file or import it but get the same error:

```
Parse error (2) at line 65, column 57:
Expected '>' or '/', but got '[a-zA-Z]'.
/home/name/Documents/Fritzing/parts/contrib/SparkFun_BadgerStick.fzp
```

bboyho, 2023-10-09:
Hi!

I just noticed this issue.  Hmm, I haven't encountered that error before? I have _Fritzing Version 0.9.3_ installed on my computer. 

Are you trying to use the BadgerStick that was included in Fritzing software already? It sounds like you are trying to import the one from this Fritzing repo? What version are you using?

I tried importing the FZPZ in my version of Fritzing but it wouldn't let me since it was already included. I tried searching online and the Fritzing forums but I couldn't find the exact reasoning behind this error. Looking briefly into the core parts (i.e. C:\...\fritzing.0.9.3b.64.pc\fritzing-parts\core) under the software for the SparkFun_BadgerStick.fzp file, it seems like there was no space between the two attributes? I'm wondering if that is causing an issue?

Nriver, 2024-12-13:
This issue is caused by the invisible space `<0x00>`. You can see it with Sublime Text.

[image]

[original - 13314_sfe_digital_temperature_sensor_breakout-TMP102.fzpz.zip](https://github.com/user-attachments/files/18123004/original.-.13314_sfe_digital_temperature_sensor_breakout-TMP102.fzpz.zip)

I manually open the zip file, remove the `<0x00>` and repack the file, it's working now.

Github only allow `.zip` file, rename the `.fzpz.zip` to `.fzpz` and import it.
[fixed - 13314_sfe_digital_temperature_sensor_breakout-TMP102.fzpz.zip](https://github.com/user-attachments/files/18123007/fixed.-.13314_sfe_digital_temperature_sensor_breakout-TMP102.fzpz.zip)

[image]

Nriver, 2024-12-13:
Another error is caused by missing space.

[image]

Nriver, 2024-12-13:
I've made a PR to fix this #35.